mwrnd/innova2_flex_xcku15p_notes #1: Have you used the Innova2 FPGA to control the network chip ingress egress packets flow?

ttsiodras, 2022-12-02:
Hi.

I saw your interesting projects on the Innova2 - and wanted to ask you:
Have you used the Innova2 FPGA to control the network chip's (ConnectX-5) ingress/egress packets?

I want to hack something that allows me to send pkts directly from the FPGA, but can't find any example for how to do this.

Sorry for opening a non-issue here - but I couldn't find any other way to communicate with you.

mwrnd, 2022-12-02:
This is a valid issue I have been exploring as well.

The Innova-2 shows up under Linux as being composed of a bridge connecting two ethernet controllers and the FPGA to the host.
[image]

The [Innova-2 Product Brief](https://www.mellanox.com/files/doc-2020/pb-innova-2-flex.pdf) specifically states it has a PCIe switch, which would allow FPGA to ConnectX-5 communication.

A team associated with Nvidia Networking has the [FlexDriver Project](https://haggaie.github.io/files/flexdriver-preprint-asplos22.pdf) which can supposedly do direct NIC-to-FPGA Bump-In-The-Wire processing. However, to make use of it you must "acquire FlexDriver IP (`src/flc.dcp`) from NVIDIA Networking". It appears to be a proprietary feature. I looked through the two FlexDriver demos, the [flexdriver-zuc encryption demo](https://github.com/acsl-technion/flexdriver-zuc) and the [flexdriver-iot-auth packet processing offload demo](https://github.com/acsl-technion/flexdriver-iot-auth) but the code to communicate with the ConnectX-5 seems to be in `src/flc.dcp`.

It is possible to [disconnect the Innova-2 from PCIe](https://github.com/mwrnd/innova2_flex_xcku15p_notes#disconnect-the-innova-2-fpga-from-the-pcie-bridge) using `setpci`:
```
sudo setpci  -s 02:08.0  0x70.w=0x50
```

[image]

Then reconnect it with `sudo setpci  -s 02:08.0  0x70.w=0x40`. Some combination of bits that connect the FPGA to the ConnectX-5 probably exists.

I took a break from playing with networking to get [RISC-V working on the Innova-2](https://github.com/mwrnd/innova2_experiments/tree/main/riscv_rocket64b4l2w_xdma).

We can try coordinating messages to the [Nvidia Networking Forums](https://forums.developer.nvidia.com/search?q=innova-2). Or, if you have an in at Nvidia, please post anything you discover and are allowed to forward.

ttsiodras, 2022-12-02:
Thanks for your feedback, much appreciated!

If you post a question about this in any NVIDIA forum, ping me here and I will join in.

In the meantime I will continue looking - and keep you posted if I find anything.

mwrnd, 2022-12-02:
I have asked about [ConnectX-5 Ethernet to FPGA Direct Communication](https://forums.developer.nvidia.com/t/innova-2-flex-connectx-5-ethernet-to-fpga-direct-communication/236059) on the [Nvidia SoC and SmartNIC Forum](https://forums.developer.nvidia.com/c/infrastructure/software-and-drivers/soc-and-smartnic/530).

I have also created GitHub Issues on the [flexdriver-iot-auth](https://github.com/acsl-technion/flexdriver-iot-auth/issues/1) and [flexdriver-zuc](https://github.com/acsl-technion/flexdriver-zuc/issues/1) projects.

mwrnd, 2022-12-07:
Seems unlikely Nvidia will provide any support for this but I did some research and came across some hopeful notes.

[FlexDriver](https://marksilberstein.com/wp-content/uploads/2021/11/flexdriver-asplos22-slides.pdf) uses [PCIe Peer-to-Peer](https://www.ni.com/en-ca/support/documentation/supplemental/10/an-introduction-to-peer-to-peer-streaming.html):

[image]

PCIe Peer-to-Peer is [supported by the ConnectX-5](https://network.nvidia.com/files/doc-2020/pb-connectx-5-en-card.pdf):

[image]

Xilinx supports PCIe Peer-to-Peer via [XRT](https://xilinx.github.io/XRT/master/html/p2p.html). Check out the Vitis [p2p_bandwidth](https://github.com/Xilinx/Vitis_Accel_Examples/tree/master/host/p2p_bandwidth) and 
[p2p_simple](https://github.com/Xilinx/Vitis_Accel_Examples/tree/master/host/p2p_simple) demos.

Mellanox [demonstrated RDMA-to-PCIe](https://www.businesswire.com/news/home/20150811005497/en/PMC-and-Mellanox-Demonstrate-NVMe-over-RDMA-and-Peer-to-Peer-Transactions-Between-RDMA-and-PCIe-Devices) and have the [nv_peer_memory](https://github.com/Mellanox/nv_peer_memory) project for GPU-to-InfiniBand Peer-to-Peer.

Unless there is some secret register that needs to be set to enable PCIe Peer-to-Peer in the ConnectX-5 PCIe switch and Ethernet Controllers, recreating FlexDriver should be possible. Or, it may be that the root complex or [motherboard PCIe switch needs to support](https://support.xilinx.com/s/question/0D52E00006hpcxVSAQ/the-mystery-of-peertopeer-transfer?language=en_US) Peer-to-Peer. In any case, this seems like a multi-month project.

More references: [XDMA Peer-to-Peer question](https://support.xilinx.com/s/question/0D52E00006hpXoiSAE/xilinx-pcie-xdma-peer-to-peer-dmabridge-subsystem-for-pci-express?language=en_US), [PCIe Performance Tuning](https://docs.xilinx.com/v/u/en-US/wp350), some [ConnectX PCIe Registers](https://support.mellanox.com/s/article/understanding-pcie-configuration-for-maximum-performance), [Ethernet to NUMA Node Mapping](https://support.mellanox.com/s/article/understanding-numa-node-for-performance-benchmarks), [Linux PCIe Peer-to-Peer DMA Support](https://www.eideticom.com/media-news/blog/33-p2pdma-in-linux-kernel-4-20-rc1-is-here.html).

If you need a solution sooner rather than later there are boards with direct connections between an FPGA and SFP28/QSFP modules. The [Alveo](https://www.xilinx.com/products/boards-and-kits/accelerator-cards.html) [U25N](https://www.xilinx.com/products/boards-and-kits/alveo/u25n.html) or [U200](https://www.xilinx.com/products/boards-and-kits/alveo/u200.html) ([DigiKey](https://www.digikey.com/en/products/detail/amd-xilinx/A-U200-A64G-PQ-G/9645682)/[eBay](https://www.ebay.com/sch/i.html?_from=R40&_trksid=m570.l1313&_nkw=alveo+u200&_sacat=0)) or [U55C](https://www.xilinx.com/products/boards-and-kits/alveo/u55c.html) ([DigiKey](https://www.digikey.com/en/products/detail/amd-xilinx/A-U55C-P00G-PQ-G/15974870)) are most similar to the Innova-2 depending on your RAM requirements. If you require an OpenCAPI connector, look into the Alpha-Data [ADM-PCIE-9V3](https://www.alpha-data.com/product/adm-pcie-9v3/) and [ADM-PCIE-9V5](https://www.alpha-data.com/product/adm-pcie-9v5/). The [corundum](https://github.com/corundum/corundum) project supports the U200 and ADM-PCIE-9V3. The Innova-2 at around [$200](https://web.archive.org/web/20221206130700/https://www.ebay.com/itm/295377434248) is great value if you want to experiment with PCIe and FPGAs but Nvidia does not seem too keen to support it. It is based on Another coMpany's proDuct and they seem focused on their [BlueField DPU](https://www.nvidia.com/en-us/networking/products/data-processing-unit/) accelerators.

---

The main MT28800 PCIe switch hosts two PCIe bridges/switches, one for the FPGA and the other for the dual MT27800 Ethernet Controllers.

[image]

None of the Mellanox-specific registers seem related to PCIe Peer-to-Peer. The following dumps all of them:
```
sudo mst start
sudo mst status
sudo mlxreg -d /dev/mst/mt4119_pciconf0 --show_regs | cut -d " " -f 1 | grep "^[A-Z]" | grep -v "Available" | xargs -t -L 1 -I{} sudo mlxreg -d /dev/mst/mt4119_pciconf0 --show_reg {}
```

I tried setting PCIe registers to 0xFF and to 0x00 and was still able to run `iperf3` between the Ethernet interfaces. There does not appear to be a simple PCIe routing or disconnect feature for Ethernet similar to the FPGA disconnect.

```
#!/bin/sh

DEV="02:08.0"

sudo lspci -s ${DEV} -vvxxxx >pciregs0

for ((i=70; i<=303; i++))
do
    cur=$(printf '%#x' $i)
    cmd="sudo setpci -s ${DEV} $cur.b=0xFF"
    echo $cmd
    $cmd
done

sudo lspci -s ${DEV} -vvxxxx >pciregs1
```

Make sure to have a way to restore all the PCIe registers as some settings survive a reboot and mess up PCIe communication.
```
DEV="02:08.0"

sudo setpci -s ${DEV}  0x60.b=0x10 ; sudo setpci -s ${DEV}  0x61.b=0xE0
sudo setpci -s ${DEV}  0x62.b=0x62 ; sudo setpci -s ${DEV}  0x64.b=0x02
sudo setpci -s ${DEV}  0x65.b=0x80 ; sudo setpci -s ${DEV}  0x68.b=0x20
sudo setpci -s ${DEV}  0x69.b=0x20 ; sudo setpci -s ${DEV}  0x6A.b=0x09
sudo setpci -s ${DEV}  0x6C.b=0x84 ; sudo setpci -s ${DEV}  0x6D.b=0xF0
sudo setpci -s ${DEV}  0x6E.b=0x5B ; sudo setpci -s ${DEV}  0x6F.b=0x08
sudo setpci -s ${DEV}  0x70.b=0x40 ; sudo setpci -s ${DEV}  0x72.b=0x83
sudo setpci -s ${DEV}  0x73.b=0x30 ; sudo setpci -s ${DEV}  0x7A.b=0x40
sudo setpci -s ${DEV}  0x84.b=0x20 ; sudo setpci -s ${DEV}  0x86.b=0x01
[... cut ...]
sudo setpci -s ${DEV}  0x10D.b=0x20 ; sudo setpci -s ${DEV}  0x10E.b=0x06
sudo setpci -s ${DEV}  0x111.b=0x20 ; sudo setpci -s ${DEV}  0x114.b=0x00
sudo setpci -s ${DEV}  0x115.b=0x20 ; sudo setpci -s ${DEV}  0x118.b=0xA0
sudo setpci -s ${DEV}  0x119.b=0x00 ; sudo setpci -s ${DEV}  0x11C.b=0xD0
sudo setpci -s ${DEV}  0x11D.b=0x3C ; sudo setpci -s ${DEV}  0x11E.b=0x7D
sudo setpci -s ${DEV}  0x11F.b=0x08 ; sudo setpci -s ${DEV}  0x120.b=0xC7
sudo setpci -s ${DEV}  0x121.b=0x7C ; sudo setpci -s ${DEV}  0x122.b=0x53
sudo setpci -s ${DEV}  0x124.b=0x08
```

ttsiodras, 2022-12-08:
Many thanks for your valuable input! I will investigate and see what I can do.

mwrnd, 2022-12-16:
[PCIe Peer-to-Peer DMA](https://www.kernel.org/doc/html/v5.8/driver-api/pci/p2pdma.html) should work on any motherboard with [Resizable BAR](https://xilinx.github.io/Vitis_Accel_Examples/2020.2/html/p2p_fpga2fpga.html) and [Above-4G Memory Decoding](https://github.com/Xilinx/XRT/blob/master/src/runtime_src/doc/toc/p2p.rst#bios-setup) [capabilities](https://www.nvidia.com/en-us/geforce/news/geforce-rtx-30-series-resizable-bar-support/). However, [it is supposedly buggy](https://docs.nvidia.com/cuda/gpudirect-rdma/index.html#supported-systems).

[GPUDirect](https://developer.nvidia.com/gpudirect) is supported on [Keplar](https://en.wikipedia.org/wiki/Kepler_(microarchitecture)) or later [Tesla and Quadro GPUs](https://docs.nvidia.com/cuda/gpudirect-rdma/index.html) using GPU [driver version R470 or later](https://github.com/Mellanox/nv_peer_memory#general).

Testing the [nv_peer_memory](https://github.com/Mellanox/nv_peer_memory#general) project would give some confidence that this is at all possible but the PCIe Peer-to-Peer communication would occur within the motherboard's [Root Complex](https://en.wikipedia.org/wiki/Root_complex) or a [PCIe switch](https://docs.broadcom.com/doc/12351852).

As I understand the [FlexDriver](https://marksilberstein.com/wp-content/uploads/2021/11/flexdriver-asplos22-slides.pdf) project, PCIe Peer-to-Peer occurs within the [ConnectX-5](https://network.nvidia.com/files/doc-2020/pb-connectx-5-en-card.pdf)'s bridge/switch.

That means the bigger hurdle is enabling [PCIe Peer-to-Peer within the FPGA](https://xilinx.github.io/XRT/master/html/p2p.html) which requires a [Vitis DFX Shell](https://xilinx.github.io/XRT/master/html/platforms.html#pcie-based-platforms). The [Alveo U200](https://www.xilinx.com/products/boards-and-kits/alveo/u200.html) is [known to support PCIe P2P](https://github.com/Xilinx/Vitis_Accel_Examples/tree/master/host/p2p_fpga2fpga#disable-p2p). It would be great if I could just try re-targeting the U200 [Deployment Target Platform](https://www.xilinx.com/products/boards-and-kits/alveo/u200.html#gettingStarted) but unfortunately the [Development Target Platform](https://www.xilinx.com/member/forms/download/eula-xef.html?filename=xilinx-u200-gen3x16-xdma-2-202110-1-dev_1-3514848_all.deb) comes as an [XSA](https://docs.xilinx.com/r/en-US/ug1400-vitis-embedded/Creating-a-Hardware-Design-XSA-File) and will [require some effort to attempt to use](https://ohwr.org/project/soc-course/wikis/Reverse-Engineering-the-XSA-File).

I have found two Vitis shells with some source code, Xilinx's [open-nic-shell](https://github.com/Xilinx/open-nic-shell) and [U200 to AWS-F1](https://github.com/aws/aws-fpga-f1-u200). They [may support](https://github.com/Xilinx/open-nic-shell/search?p=2&q=p2p) PCIe P2P.

Another avenue to explore is [PCIe P2P between GPU and FPGA](https://forums.developer.nvidia.com/t/p2p-communication-between-gpu-and-fpga/219778) which seems more mature but [there is no official support](https://github.com/Xilinx/XRT/issues/6618).

Again, if you are working on a near-term project, the [Alveo U200](https://www.xilinx.com/products/boards-and-kits/alveo/u200.html) ([DigiKey](https://www.digikey.com/en/products/detail/amd-xilinx/A-U200-A64G-PQ-G/9645682)/[eBay](https://www.ebay.com/sch/i.html?_from=R40&_trksid=m570.l1313&_nkw=alveo+u200&_sacat=0)) has two QSFP28 (100GbE) cages connected directly to an [`XCU200 = XCVU9P`](https://en.wikipedia.org/wiki/List_of_Xilinx_FPGAs#UltraScale_and_UltraScale+) FPGA. You could then use [100G QSFP28 to 4x25G SFP28 Direct Attach Breakout Cables](https://www.fs.com/products/74665.html?attribute=1439&id=515834) to get up to 8 25GbE interfaces communicating through Programmable Logic.

mwrnd, 2023-01-06:
I have made progress figuring out what **doesn't** work. I got a [Tesla K80](https://www.nvidia.com/en-gb/data-center/tesla-k80/) GPU ([SM_3.7](https://developer.nvidia.com/cuda-gpus)) and have been trying to demonstrate [P2PDMA](https://www.kernel.org/doc/html/latest/driver-api/pci/p2pdma.html). The K80 is [no longer under active development](https://forums.developer.nvidia.com/t/tesla-k80-and-discontinued-driver-support-for-kepler/193903) so I am stuck in a version and dependency nightmare. [My setup notes](https://github.com/mwrnd/innova2_experiments/tree/main/innova2_tesla-k80_pcie_p2pdma).

For example, [`gdscheck.py`](https://docs.nvidia.com/gpudirect-storage/troubleshooting-guide/index.html#verify-suc-install) is part of [CUDA 11.4](https://developer.nvidia.com/cuda-11-4-4-download-archive) (last version to officially support the K80) and it asks for [CUDA Compute Capability](https://developer.nvidia.com/cuda-gpus) >=6 which is Pascal or later. DPDK only officially supports [Ampere, Turing, and Volta (SM_7+)](https://doc.dpdk.org/guides/gpus/cuda.html#supported-gpus).

Motherboard, GPU, NIC, and software all have to work together and thus far I am only convinced the GPU I am using supports PCIe P2P ([PEX 8747](https://docs.broadcom.com/doc/12351854)). I intend to spend some more time with the software. A known-working server or [PCIe Switch Card](https://web.archive.org/web/20230103002331/https://www.ebay.com/itm/134368117447) is too much right now. I am using an [Intel B360](https://www.intel.com/content/dam/www/public/us/en/documents/product-briefs/b360-chipset-product-brief.pdf) motherboard which supposedly supports P2P (search for `peer` in the [datasheet](https://www.intel.com/content/dam/www/public/us/en/documents/technical-specifications/300-series-chipset-on-package-pch-datasheet-vol-2.pdf)).

I managed to get [open-nic-shell](https://github.com/Xilinx/open-nic-shell) to synthesize and implement but it does not show up on the PCIe bus. I believe this is an issue with Vivado 2021.2 as I have been unable to get even the simplest QDMA demo to work with the XCKU15P. I will try again.

[aws-fpga](https://github.com/awslabs/aws-fpga) is unfortunately unusable with anything but the [AWS F1](https://aws.amazon.com/ec2/instance-types/f1/) instances and the [Alveo U200](https://www.xilinx.com/products/boards-and-kits/alveo/u200.html) as the core shell comes as an encrypted [Vivado checkpoint](https://docs.xilinx.com/r/en-US/ug892-vivado-design-flows-overview/Using-Design-Checkpoints). However, it may still be useful to spin up an F1 instance and run the [cl_dram_dma](https://github.com/aws/aws-FPGA/blob/master/hdk/cl/examples/cl_dram_dma/README.md) demo to see how [P2PDMA looks and functions](https://github.com/awslabs/aws-fpga-app-notes/tree/master/Using-PCIe-Peer2Peer) on a working setup.

mwrnd, 2023-04-30:
There are two [CMAC 100G MAC](https://docs.xilinx.com/r/en-US/ug974-vivado-ultrascale-libraries/CMAC) blocks adjacent to the OpenCAPI GTY Quads 131 and 132 so it should be possible to implement dual QSFP modules using the SlimSAS OpenCAPI connector. Refer to Figure 1-99 on Pg#134 of [ug575 UltraScale+ Packaging and Pinouts](https://docs.xilinx.com/v/u/en-US/ug575-ultrascale-pkg-pinout).

Unfortunately I am having problems getting my [PCIe-to-SlimSAS Adapter](https://github.com/mwrnd/PCIe-to-SlimSAS8x_Adapter) working. The [OpenPower AAC](https://files.openpower.foundation/s/xSQPe6ypoakKQdq/download/25Gbps-spec-20171108.pdf) specification I used has a different pinout to the [ADM-PCIe-9V5](https://www.alpha-data.com/xml/user_manuals/adm-pcie-9v5%20user%20manual_v1_4.pdf) which is the only OpenCAPI pinout I have seen.

I mangled and reworked my PCIe-to-SlimSAS Adapter but am unable to get PCIe to work even at Gen1 x1. Clock and Reset work well and I can successfully run [IBERT](https://docs.xilinx.com/v/u/en-US/pg246-in-system-ibert) and see a decent eye diagram on the RX channel but nothing on TX. I believe the TX pins are way off on the adapter. I intend to design a SlimSAS breakout to figure out the signal locations. That or my board has broken OpenCAPI transceivers.

[image]

If I manage to get the PCIe Adapter working I will go ahead and complete a [SlimSAS-to-QSFP Adapter](https://github.com/mwrnd/SlimSAS8x-to-QSFP_Adapter).

mwrnd, 2023-11-13:
@ciclonite which version of the board do you have?

[image]

ciclonite, 2023-11-14:
> @ciclonite which version of the board do you have (what is the DDR4 FBGA Code)?
The six board that i've unpacked have these chip:

[image]

Tomorrow i'll look on the other. 

Great! you are doing an immense work!
I've write to nvidia for the FlexDriver IP but haven't reply.