mwrnd/innova2_flex_xcku15p_notes #3: Add Bitstreams and Files for the 4GB Version of the MLNX Innova 2 Flex

CuteSC2, 2023-04-30:
There exist a 4GB version of the Innova 2 Flex with MT40A512M16 DDR4 ICs.
So there should be Bitstreams, Documentation and Project files provided.

- [ ] Document the difference of the physical board.
- [ ] Add Technical documentation.
- [ ] Provide pin mappings
- [ ] Adapt the Projects TCL scripts to take a 4GB model into account.
- [ ] Add Bitstreams for the 4GB Model

Applepi, 2023-07-31:
It looks like I have 3 of the 4GB non-crypto version of the card coming in the mail from ebay, when I get home I'll setup an environment to test these. I have a plan to use these with an external TB3 PCI-E riser on MacOS since Mellanox drivers were added in Ventura but I'll setup using the base environment first to kick the tires.

mwrnd, 2023-08-09:
I got one of the 4GB MNV303212A-ADIT boards, Rev:A2, which is the same revision as my 8GB board. The [1-byte wide DDR4 test project](https://github.com/mwrnd/innova2_ddr4_troubleshooting/tree/main/test_adit_mt40a512m16) works with `MT40A512M16JY-075` as the DDR4 component! I will create a design to test the full 4GB.

However, there are other issues with the 4GB AD**I**T boards compared to the 8GB AD**L**T. The **PSID** is `MT_0000000142` for the 4GB AD**I**T. The 8GB AD**L**T has `PSID: MT_0000000158`. I tested every version of [mlxup](https://network.nvidia.com/support/firmware/mlxup-mft/) from **4.10.0** to **4.24.0** and *none* have any firmware update for the 4GB AD**I**T. I saved and then [erased the ConnectX-5 Firmware Flash IC](https://github.com/mwrnd/innova2_flex_xcku15p_notes#programming-the-connectx5-flash-ic) and programmed it with the [8GB AD**L**T `MORSE_FW` firmware](http://www.mellanox.com/downloads/fpga/flex/Innova_2_Flex_Open_18_12.tar.gz). The FPGA+DDR4 and the PCIe switch within the ConnectX-5 seem to be working after some testing.

Only one of the 25GbE interfaces shows up in `sudo lspci -vnn` while running the original firmware. Both show up when it is programmed with the 8GB AD**L**T firmware.
```
03:00.0 Class [2000]: Mellanox Technologies Innova-2 Flex Burn image [15b3:0264]
	Subsystem: Mellanox Technologies Innova-2 Flex Burn image [15b3:0264]
	Flags: fast devsel
	Memory at 4016000000 (64-bit, prefetchable) [size=2M]
	Capabilities: [40] Power Management version 3
	Capabilities: [48] MSI: Enable- Count=1/1 Maskable- 64bit+
	Capabilities: [70] Express Endpoint, MSI 00
	Capabilities: [100] Advanced Error Reporting
	Capabilities: [180] Alternative Routing-ID Interpretation (ARI)
	Capabilities: [1c0] Secondary PCI Express
	Kernel driver in use: mlx_fpga_bope

04:00.0 Ethernet controller [0200]: Mellanox Technologies MT27800 Family [ConnectX-5] [15b3:1017]
	Subsystem: Mellanox Technologies MT27800 Family [ConnectX-5] [15b3:0043]
	Flags: fast devsel
	Memory at 4010000000 (64-bit, prefetchable) [size=32M]
	Capabilities: [60] Express Endpoint, MSI 00
	Capabilities: [48] Vital Product Data
	Capabilities: [9c] MSI-X: Enable- Count=64 Masked-
	Capabilities: [c0] Vendor Specific Information: Len=18 <?>
	Capabilities: [40] Power Management version 3
	Capabilities: [100] Advanced Error Reporting
	Capabilities: [150] Alternative Routing-ID Interpretation (ARI)
	Capabilities: [180] Single Root I/O Virtualization (SR-IOV)
	Capabilities: [1c0] Secondary PCI Express
	Kernel modules: mlx5_core
```

Applepi, 2023-08-20:
I can confirm as well that when I flash the 8GB ADLT firmware both interfaces show up but I'm unable to get the card to detect optics on either firmware. And on my 12900K machine when it trys to boot Linux it crashes and reboots.

Applepi, 2023-08-20:
I think these cards may be using a different chip entirely from the other ADLT models, there was a document that I found referring to this card as a Innova-2 Flex P^2 programmable pipeline. It was also saying that it was an EN card so maybe this card is using a different chip?  But at the same time the document states it should support OFED. But I've tried MLNX EN and OFED and inbox drivers and I've not been able to get a link up on any OS with these cards. I only have a few more days before my return window on these cards close but I may just return them if I cannot use the NIC side of these cards they are kinda useless to me. I have REV3 Engineering sample cards that show up as `MT_0000000141`  `MNV303212A-ADAT_C18_Ax` `Innova-2 Programmable EN Adapter; dual SFP port; 25GbE; PCIe3/4 x8; HHHL; active heat sink; tall bracket; ROHS R6` there is no firmware available through MLXUP.

I also tried flashing the lenovo firmware because they had referred to the cards as Innova-2 Flex EN in some cases but it just put a signed firmware on the card with the same issues lol.  

https://indico.psi.ch/event/6393/sessions/3484/attachments/11629/14919/Mellanox_PSI.pdf

https://manualzz.com/doc/33751280/innova%E2%84%A2-2-programmable-adapter-card

Applepi, 2023-08-22:
I was able to get the cards working on non-12th Z690 machines, my Z87 and X570 boards worked just fine on the 16.22.0264 firmware. Only one link worked, they established a link at 40gbps (a bug maybe) but max speed in iperf3 was only 25gbps. 

I only needed the dual 25GbE on my Z690 NAS PC so I'll likely just switch that with a quad port broadcom NIC. I will likely keep the cards even though they only establish a single link the FPGA side is super useful. I think there is some sort of issue with the 12th gen PCI-E controller and the Mellanox NIC because both my Z690 12900k and and Z690 12700k machines cannot properly initialize the NICs, I've not been able to find anyone else with this issue so it may be an issue between the old firmware and modern UEFIs.

mwrnd, 2023-08-24:
@Applepi thanks for finding the [4GB MNV303212A-ADAT Product Brief](https://manualzz.com/download/33751280)(_manualzz.com_). The following diagram from it shows a direct connection between an SFP cage and the FPGA. I have seen this diagram before but thought it was a logical description of how the board can be used as it does not describe the 8GB ADLT boards. However, it may be a physical connection diagram for the ADAT boards. That would make sense why my 4GB ADIT board with its original firmware only has one Ethernet interface. The second connects to the FPGA.

[image]

From the Mellanox presentation (_Mellanox_PSI.pdf_) the ADAT board is designed for pipeline packet processing. First the FPGA processes packets then the ConnectX-5.

[image]

The [8GB MNV303212A-ADLT Product Brief](https://network.nvidia.com/files/doc-2020/pb-innova-2-flex.pdf) shows both 25GbE connected to the ConnectX-5.

[image]

The 4GB MNV303212A-ADAT Product Brief also mentions it specifically supports "bump-in-the-wire" packet processing. It also states a Board Support Package (BSP) for the FPGA is available but I have little hope of finding it.

[image]

[Seeing as how the ADAT/ADIT EoL Notification](https://network.nvidia.com/files/pdf/eol/LCR-000437.pdf) lumps the two boards together and given all the above, it is worthwhile to check if there is a direct connection between an SFP cage and the FPGA on the 4GB ADIT boards. If yes, it should be possible to port [corundum](https://github.com/corundum/corundum) to it which would turn it into a single-port 25GbE PCIe Ethernet Adapter.

@Applepi, do you happen to have an [SFP Loopback](https://cablesondemand.com/img/product/SF-SFPPLOOPBK.pdf)([DigiKey](https://www.digikey.com/en/products/detail/amphenol-cables-on-demand/SF-SFPPLOOPBK-003-5/6873737)) and a [Xilinx-Compatible JTAG Adapter](https://docs.xilinx.com/r/en-US/ug908-vivado-programming-debugging/JTAG-Cables-and-Devices-Supported-by-hw_server) or [clone](https://www.waveshare.com/platform-cable-usb.htm)? We could try using the FPGA's GT Transceiver built-in test interface to look for the pins connected to the SFP Cage.

mwrnd, 2023-08-24:
> I was able to get the cards working on non-12th Z690 machines

It fails on my [Intel 300-Series](https://ark.intel.com/content/www/us/en/ark/products/series/126380/intel-300-series-desktop-chipsets.html) motherboard.

Trying to [disable cryptographic](https://github.com/mwrnd/innova2_flex_xcku15p_notes/issues/3#issuecomment-1688828212) functions on the 4GB ADIT did not work. It does NOT have any related registers. If curious, here are the `mlxreg` register listings for the [MNV303212A-AD**I**T](https://gist.github.com/mwrnd/d732eb8efdaf385f67314b000a4df01c) and the [MNV303212A-AD**L**T](https://gist.github.com/mwrnd/a7bd6d21a08be1a1123da77959aaef71)

The **16.22.1002** firmware does NOT work with `innova2_flex_app` from [Innova_2_Flex_Open_18_12.tar.gz](http://www.mellanox.com/downloads/fpga/flex/Innova_2_Flex_Open_18_12.tar.gz). I needed to update the firmware to the 8GB ADLT firmware to make use of the FPGA.

`sudo ~/Innova_2_Flex_Open_18_12/app/innova2_flex_app -v`
```
===============================================
 Verbosity:        1
 BOPE device:      None
 ConnectX device:  None
Cannot find appropriate ConnectX device
```

The PCIe Device Tree shows only one 25GbE interface sitting on the ConnectX-5's PCIe switch.

`sudo lspci -vnn -t`
```
-[0000:00]-+-00.0  Intel Corporation Device [8086:3e0f]
           +-1d.0-[01-04]----00.0-[02-04]--+-08.0-[03]----00.0  Mellanox Technologies Innova-2 Flex Burn image [15b3:0264]
           |                               \-10.0-[04]----00.0  Mellanox Technologies MT27800 Family [ConnectX-5] [15b3:1017]
```

`sudo lspci -vvvnnxxx -d 15b3:`

```
01:00.0 PCI bridge [0604]: Mellanox Technologies MT28800 Family [ConnectX-5 PCIe Bridge] [15b3:1974] (prog-if 00 [Normal decode])
	Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 0, Cache Line Size: 64 bytes
	Bus: primary=01, secondary=02, subordinate=04, sec-latency=0
	I/O behind bridge: 0000f000-00000fff [disabled]
	Memory behind bridge: fff00000-000fffff [disabled]
	Prefetchable memory behind bridge: 00000000a0000000-00000000a7ffffff [size=128M]
	Secondary status: 66MHz- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- <SERR- <PERR-
	BridgeCtl: Parity- SERR+ NoISA- VGA- VGA16- MAbort- >Reset- FastB2B-
		PriDiscTmr- SecDiscTmr- DiscTmrStat- DiscTmrSERREn-
	Capabilities: [60] Express (v2) Upstream Port, MSI 00
		DevCap:	MaxPayload 512 bytes, PhantFunc 0
			ExtTag- AttnBtn- AttnInd- PwrInd- RBE+ SlotPowerLimit 25.000W
		DevCtl:	CorrErr- NonFatalErr- FatalErr- UnsupReq-
			RlxdOrd- ExtTag- PhantFunc- AuxPwr- NoSnoop-
			MaxPayload 256 bytes, MaxReadReq 512 bytes
		DevSta:	CorrErr+ NonFatalErr- FatalErr- UnsupReq+ AuxPwr- TransPend-
		LnkCap:	Port #0, Speed 8GT/s, Width x8, ASPM not supported
			ClockPM- Surprise- LLActRep- BwNot- ASPMOptComp+
		LnkCtl:	ASPM Disabled; Disabled- CommClk+
			ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
		LnkSta:	Speed 8GT/s (ok), Width x4 (downgraded)
			TrErr- Train- SlotClk+ DLActive- BWMgmt- ABWMgmt-
		DevCap2: Completion Timeout: Not Supported, TimeoutDis-, NROPrPrP-, LTR-
			 10BitTagComp+, 10BitTagReq-, OBFF Not Supported, ExtFmt-, EETLPPrefix-
			 EmergencyPowerReduction Not Supported, EmergencyPowerReductionInit-
			 FRS-
			 AtomicOpsCap: Routing-
		DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis-, LTR-, OBFF Disabled
			 AtomicOpsCtl: EgressBlck-
		LnkCtl2: Target Link Speed: 8GT/s, EnterCompliance- SpeedDis-
			 Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
			 Compliance De-emphasis: -6dB
		LnkSta2: Current De-emphasis Level: -6dB, EqualizationComplete+, EqualizationPhase1+
			 EqualizationPhase2+, EqualizationPhase3+, LinkEqualizationRequest-
	Capabilities: [e0] MSI: Enable- Count=1/4 Maskable- 64bit+
		Address: 0000000000000000  Data: 0000
	Capabilities: [40] Power Management version 3
		Flags: PMEClk- DSI- D1- D2- AuxCurrent=375mA PME(D0-,D1-,D2-,D3hot-,D3cold+)
		Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=0 PME-
[... 314 lines cut ...]
		VF offset: 1, stride: 1, Device ID: 1018
		Supported Page Size: 000007ff, System Page Size: 00000001
		Region 0: Memory at 00000000a2000000 (64-bit, prefetchable)
		VF Migration: offset: 00000000, BIR: 0
	Capabilities: [1c0 v1] Secondary PCI Express
		LnkCtl3: LnkEquIntrruptEn-, PerformEqu-
		LaneErrStat: 0
	Kernel modules: mlx5_core
00: b3 15 17 10 02 00 10 00 00 00 00 02 10 00 00 00
10: 0c 00 00 a0 00 00 00 00 00 00 00 00 00 00 00 00
20: 00 00 00 00 00 00 00 00 00 00 00 00 b3 15 43 00
30: 00 00 00 00 60 00 00 00 00 00 00 00 ff 00 00 00
40: 01 00 c3 81 08 00 00 00 03 9c 38 81 00 78 00 00
50: 00 00 00 00 00 00 00 00 14 00 0f 00 0d 02 00 00
60: 10 48 02 00 e2 8f 00 10 30 29 00 00 03 f1 43 00
70: 00 00 03 11 00 00 00 00 00 00 00 00 00 00 00 00
80: 00 00 00 00 1f 00 00 00 00 00 00 00 0e 00 00 00
90: 03 00 1e 00 00 00 00 00 00 00 00 00 11 c0 3f 00
a0: 00 20 00 00 00 30 00 00 00 00 00 00 00 00 00 00
b0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
c0: 09 40 18 00 0a 00 00 20 eb 34 00 00 00 00 00 00
d0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
e0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
f0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
```

Applepi, 2023-08-28:
A thought that just occurred to me that if the ADIT / ADAT cards are advertised as a bump in the wire / crypto cards that they might have the 2nd port of the mellanox NIC connected directly to the FPGA but I'm not sure how we would verify that as we haven't been able to get any dual port firmware to get the CX-5 device to establish link.

mwrnd, 2023-08-30:
>ADIT / ADAT cards are advertised as a bump in the wire / crypto cards that they
>might have the 2nd port of the mellanox NIC connected directly to the FPGA

Yes, the ADAT [Architecture Diagram](https://user-images.githubusercontent.com/38073881/262849376-fd32fe9f-5507-4582-ba35-fe67c54a355b.png) from the [Product Brief](https://manualzz.com/download/33751280)(_manualzz.com_) may be accurate. [Gigabit Transceiver](https://en.wikipedia.org/wiki/Multi-gigabit_transceiver) signals require [inline DC-block coupling capacitors](https://en.wikipedia.org/wiki/DC_block) so the signals must be exposed somewhere on the PCB. If the ConnectX-5 could be forced to output on the TX line without an ethernet cable being detected it should be possible to find the coupling capacitors. Could also try [corundum](https://github.com/corundum/corundum) on every possible MGTREFCLK and transceiver pair on the FPGA.

Comparing the [front](https://github.com/mwrnd/innova2_flex_xcku15p_notes/blob/main/img/ADLT_vs_ADIT_Front-of-PCB_Comparison.jpg) and [back](https://github.com/mwrnd/innova2_flex_xcku15p_notes/blob/main/img/ADLT_vs_ADIT_Back-of-PCB_Comparison.jpg) of the 4GB ADIT and 8GB ADLT boards I found two oscillators populated on the 4GB ADIT board that are **not** on the ADLT. To minimize noise and jitter these are almost certainly placed as close as possible to their transceiver. @Applepi please confirm whether you see the same on the ADAT boards.

[image]

These oscillators are on the Pin1 edge of the [FFVE1517](https://docs.xilinx.com/r/en-US/ug575-ultrascale-pkg-pinout/FFVE1517-XCKU15P-and-FFRE1517-XQKU15P) package and the closest [MGTs](https://en.wikipedia.org/wiki/Multi-gigabit_transceiver) are the [GTH](https://docs.xilinx.com/v/u/en-US/ug576-ultrascale-gth-transceivers) Banks of the [XCKU15P](https://docs.xilinx.com/r/en-US/ug575-ultrascale-pkg-pinout/XCKU15P-and-XQKU15P-Bank-Diagrams). That means quite a few MGTREFCLK banks will need to be tested.

[image]

mwrnd, 2023-09-13:
The ADIT oscillators are labeled `J161.13 DFS 7L`. 161.13MHz is a [standard 10G/25G](https://support.xilinx.com/s/question/0D52E00006hpt6rSAA/gtrefclk-15625mhz-or-1611328125mhz-for-10g-ethernet-system?language=en_US) Ethernet reference clock.

[image]

> CX-5 Device it looks as though there may be 16 DIFF pairs

I can confirm that both the ADLT and ADIT have PCIe x8 connections to the ConnectX-5 which means 8 differential pairs for both TX and RX or 16 total.

Here is a close-up of the ConnectX-5 IC on the Innova2 8GB AD**L**T. There are 8 differential pairs leading to the FPGA on the right, a 156.25MHz oscillator at top, but notice no obvious coupling capacitors on the left side toward the SFP connectors.

[image]

Here is a close-up of the ConnectX-5 IC on the Innova2 4GB AD**I**T. There are 8 differential pairs leading to the FPGA on the right, a 156.25MHz oscillator at top, but notice the 4 differential pairs leading to the SFP connectors on the left.

[image]

Similar situation for the underside of the ConnectX-5 IC on the Innova2 8GB AD**L**T. 8 differential pairs on the left going from the CX5 to the FPGA but none toward the SFP connectors on the right.

[image]

The underside of the ConnectX-5 IC on the Innova2 4GB AD**I**T has 8 differential pairs on the left going from the CX5 to the FPGA but also 4 diff pairs toward the SFP connectors on the right.

[image]

@Applepi please check whether the AD**A**T has such capacitor connections to the SFP connectors.

From the [SFP Specification (INF-8074i)](https://members.snia.org/document/dl/26184), there is just one RX and one TX per module. There is a dual channel [SFP-DD Spec](http://sfp-dd.com/wp-content/uploads/2022/03/SFP-DDrev5.1.pdf) but it uses a different connector. [InfiniBand](https://en.wikipedia.org/wiki/InfiniBand) uses QSFP.

mwrnd, 2023-09-13:
I just noticed the Innova2 8GB AD**L**T uses the `MT28808A0-FCCF-EV` while the Innova2 4GB AD**I**T uses the `MT27808A0-FCCF-EV`. The updated IC may use a different coupling interface.

mwrnd, 2023-11-12:
By testing unused Quads I found that the 161.13MHz oscillators are connected to pins V27 and Y27, Quads 129 and 130.

[image]

```
# MNV303212A-ADIT and MNV303212A-ADAT Only:
# SFP - CMAC X0Y0 and X0Y1
# GTY Quad 129 == Quad_X0Y2
# AD36=MGTYRXP0=CHANNEL_X0Y8, AC38=MGTYRXP1=CHANNEL_X0Y9,
# AB36=MGTYRXP2=CHANNEL_X0Y10, AA38=MGTYRXP3=CHANNEL_X0Y11
# Y27=MGTREFCLK0_P, W29=MGTREFCLK1_P
# GTY Quad 130 == Quad_X0Y3
# AD36=MGTYRXP0=CHANNEL_X0Y12, AC38=MGTYRXP1=CHANNEL_X0Y13,
# AB36=MGTYRXP2=CHANNEL_X0Y14, AA38=MGTYRXP3=CHANNEL_X0Y15
# V27=MGTREFCLK0_P, U29=MGTREFCLK1_P

#set_property PACKAGE_PIN V27 [get_ports diff_clock_rtl_2_clk_p]
#create_clock -period 6.206 -name sys_clk [get_ports diff_clock_rtl_2_clk_p]

#set_property PACKAGE_PIN Y27 [get_ports diff_clock_rtl_3_clk_p]
#create_clock -period 6.206 -name sys_clk [get_ports diff_clock_rtl_3_clk_p]
```

I plan to test Channels 0 and 3 of the Quads to see if any work with the following loopback.

Using the pinout of the [Small Formfactor Pluggable SFP Transceiver INF-8074i Standard](https://www.gigalight.com/downloads/standards/INF-8074i.pdf):

[image]

I created a simple loopback with an [SFP Connector](https://www.digikey.com/en/products/detail/amphenol-cs-commercial-products/UE75-A20-6000T/1242770) and a [Direct-Attach Cable](https://www.fs.com/products/65841.html). Only two short [30AWG wires](https://www.digikey.com/en/products/detail/cnc-tech/10368-32-1-0500-005-1-TS/19721772)([alt](https://www.digikey.com/en/products/detail/cnc-tech/10064-30-1-0500-006-1-TS/4496900)) were needed, RXp-to-TXp and RXn-to-TXn, as all GNDs are connected together in a module or cable.

[image]

There are ~10nF=0.01uF capacitors in-line at both ends of a [Direct-Attach Cable](https://www.fs.com/products/65841.html) for [about 5nF total](https://en.wikipedia.org/wiki/Capacitor#Circuit_analysis).

[image]

mwrnd, 2024-01-08:
I got DDR4 working on my Innova2 4GB ADIT board.

@Applepi please test your ADAT board with the [innova2_4gb_adit_xdma_ddr4_demo](https://github.com/mwrnd/innova2_4gb_adit_xdma_ddr4_demo) project when you have some spare time.

mwrnd, 2024-01-26:
I implemented an [IBERT for GTY](https://docs.xilinx.com/v/u/en-US/pg196-ibert-ultrascale-gty) example design and with the SFP loopback cable plugged into the top SFP port (farthest from PCIe card edge) I seem to be getting communication between 
`X0Y10_TX` and `X0Y9_RX` of Quad 129 (MGTREFCLK0 == `Y27`).

[image]

mwrnd, 2024-02-21:
Using the DIY SFP Loopback and the [xcku15p_ffve1517_GTY_IBERT](https://github.com/mwrnd/innova2_experiments/tree/main/xcku15p_ffve1517_GTY_IBERT) project I was able to determine that the second SFP Port on the MNV303212A-ADIT is connected to Quad 129 Channel X0Y10.

[image]

With the SFP Loopback connected to SFP Port 2 (closest to PCIe card edge):

[image]

IBERT is successful:

[image]

With the SFP Loopback connected to SFP Port 1:

[image]

There are no signal links:

[image]

mwrnd, 2024-02-23:
The 4GB MNV303212A-AD**I**T has a significantly different OpenCAPI pinout than the Innova2 8GB MNV303212A-AD**L**T. This is why the [v0.1-alpha OpenCAPI_Breakout](https://github.com/mwrnd/OpenCAPI_Breakout/releases/tag/v0.1-alpha) adapter does not work with it. Refer to [hardware_notes](https://github.com/mwrnd/innova2_flex_xcku15p_notes/tree/main/hardware_notes) and the [xcku15p_ffve1517_GTY_IBERT](https://github.com/mwrnd/innova2_experiments/tree/main/xcku15p_ffve1517_GTY_IBERT) project.
```
X0Y16: RX3p-TX0n, RX3n-TX0p
X0Y17: RX0p-TX1n, RX0n-TX1p
X0Y18: RX2p-TX2p, RX2n-TX2n
X0Y20: RX4p-TX4n, RX4n-TX4p
X0Y21: RX5p-TX5p, RX5n-TX5n
X0Y22: RX6p-TX6p, RX6n-TX6n
X0Y23: RX7p-TX7p, RX7n-TX7n
```

Applepi, 2026-10-24:
I have confirmed using your IBERT and GPIO bitstreams that MIG calibration passes and we get an eye when using a loopback adapter on the ADAT card. 

Sorry for the delay in getting back to this,

[image]