mwrnd/innova2_flex_xcku15p_notes #5: mlxup not showing available online updates

yangl1996, 2023-05-02:
I tried to use `mixup` to fetch the latest firmware, but got the following output

```
Device #4:
----------

  Device Type:      ConnectX5
  Part Number:      MNV303212A-ADL_Ax
  Description:      Innova-2 Flex for Application Acceleration; ConnectX-5 Eth Adapter dual Port SFP 25GbE; Xilinx KU15P; PCI4.0 x8; HHHL; active cooling tall bracket; ROHS R6
  PSID:             MT_0000000158
  PCI Device Name:  pci0:6:0:0
  Base GUID:        N/A
  Base MAC:         N/A
  Versions:         Current        Available     
     FW             16.24.4020     N/A           

  Status:           No matching image found
```

Could it be that Nvidia has discontinued the board (the Lenovo variant is pull out from the market as of the beginning of 2023) and stopped serving the firmwares?

mwrnd, 2023-05-02:
Try running [`mlxup`](https://network.nvidia.com/support/firmware/mlxup-mft/) without the `--online` option. Both version `4.20.0` and the latest, `4.23.0`. [`sudo ./mlxup`](https://github.com/mwrnd/innova2_flex_xcku15p_notes#upgrading-the-connectx5-firmware) worked for me.
```
wget https://www.mellanox.com/downloads/firmware/mlxup/4.20.0/SFX/linux_x64/mlxup -O mlxup_4.20.0
wget https://www.mellanox.com/downloads/firmware/mlxup/4.23.0/SFX/linux_x64/mlxup -O mlxup_4.23.0
```

Innova-2 [MNV303212A-ADL ConnectX-5 Firmware 16.26.4012](http://www.mellanox.com/downloads/firmware/fw-ConnectX5-rel-16_26_4012-MNV303212A-ADL_Ax-UEFI-14.19.17-FlexBoot-3.5.805.bin.zip) is still directy downloadable as of today.

I do not have any contacts at Nvidia. I do not believe the Innova-2 was ever for sale on the (https://store.nvidia.com/en-us/networking/store/?page=1&limit=9&locale=en-us&category=ADAPTER_CARDS&search=innova)[official shop]. Not great news that Lenovo has pulled it from market.

yangl1996, 2023-05-02:
Interesting. `mlxup` was able to find the firmware I manually downloaded to the same directory (thanks for the link!), but cannot fetch the 16.28 version from the internet (with or without the `--online` option). Might be just some quirkiness of the FreeBSD-version MFT tools. Will report back if I make further progress.

mwrnd, 2023-05-02:
I had an error in my notes and have [updated them](https://github.com/mwrnd/innova2_flex_xcku15p_notes#upgrading-the-connectx5-firmware). `mlxup` **4.15.2** is the [latest version](https://network.nvidia.com/support/firmware/mlxup-mft/) to support the Innova-2 and will update ConnectX-5 firmware to **16.28.2006**. **16.24.4020** is included in the [Innova-2 package](http://www.mellanox.com/downloads/fpga/flex/Innova_2_Flex_Open_18_12.tar.gz) and (**16.26.4012** is directly downloadable)[https://network.nvidia.com/support/firmware/connectx5en/].

[image]

**4.20.0** fails:

[image]

As does the latest **4.23.0**:

[image]

yangl1996, 2023-05-02:
Thanks so much for the clarification (and the entire notes)! Will try it later this week.

(Feel free to close this issue, since the main document is updated : )

mwrnd, 2023-05-02:
Thanks for letting me know about the error in my notes.

The biggest issue with using the Innova-2 is wrangling the correct software versions and their dependencies. If you intend to make an attempt at [FPGA processing of packets using PCIe P2PDMA](https://github.com/mwrnd/innova2_flex_xcku15p_notes/issues/1), the only publicly available clue is that it [will require Vivado 2019.2](https://github.com/acsl-technion/flexdriver-zuc/blob/688c44720cd7bc13e5e05dbf0695ef95a2e64afa/ip/xilinx_pcie4/xilinx_pcie4.xci#L1226) but note the actual code is [NOT publicly available](https://github.com/acsl-technion/flexdriver-zuc/issues/1). I have been able to compile working [XDMA+DDR4 demos using Vivado 2018.3, 2020.2, and 2021.2 whereas 2017.2 and 2021.1 fail](https://github.com/mwrnd/innova2_flex_xcku15p_notes#install-vivado-or-vivado-lab-edition). I have had some tentative success compiling a QDMA demo using Vivado 2022.1.

closed 2023-05-02 by mwrnd