mwrnd/innova2_flex_xcku15p_notes #7: New firmware version found on one of my board

ciclonite, 2023-11-14:
Hi,
Non-issue here.
In one of these card, i've found this firmware : 
Image type:            FS4
**FW Version:            16.28.2006**
FW Release Date:       15.9.2020
Product Version:       16.28.2006
Rom Info:              type=UEFI version=14.21.17 cpu=AMD64
                       type=PXE version=3.6.102 cpu=AMD64
Description:           UID                GuidsNumber
Base GUID:             e41d2d0300e70f8e        4
Base MAC:              e41d2de70f8e            4
Image VSD:             N/A
Device VSD:            N/A
PSID:                  MT_0000000158
Security Attributes:   N/A

No one of the other card have this FW.
Someone have see this?

mwrnd, 2023-11-22:
Someone must have run [mlxup](https://network.nvidia.com/support/firmware/mlxup-mft/) [4.15.2](http://www.mellanox.com/downloads/firmware/mlxup/4.15.2/SFX/linux_x64/mlxup) on the card. **16.28.2006** is the latest firmware version. Refer to [Upgrading the ConnectX5 Firmware].(https://github.com/mwrnd/innova2_flex_xcku15p_notes#upgrading-the-connectx5-firmware)

[image]

AGenchev, 2024-01-12:
To document it: I updated a 4GB ADIT with it. After the update I got PXE & UEFI just like the picture above. This enabled the ADIT to recover (somebody had bricked it) and allowed flex app to work enough to allow flashing with user image, enabling jtag, etc.