mwrnd/innova2_flex_xcku15p_notes #8: ConnextX missing - should I flash?

AGenchev, 2023-11-25:
Hello, someone handed me a lenovo-made innova2. It is missing the connect-x device. 
I think there is a problem with the card, but I can't diagnose it for now.
```
sudo flint -d /dev/mst/mt4119_pciconf0 q
Image type:            FS4
FW Version:            16.24.4020
FW Release Date:       3.1.2019
Product Version:       16.24.4020
Rom Info:              type=UEFI version=14.17.13 cpu=AMD64
                       type=PXE version=3.5.603 cpu=AMD64
Description:           UID                GuidsNumber
Base GUID:             aaaaaa0300aaaaa0        12
Base MAC:              aaaaaaaaaaa0            12
Image VSD:             N/A
Device VSD:            N/A
PSID:                  LNV0000000025
Security Attributes:   secure-fw
```
```
Sudo ~/Innova_2_Flex_Open_18_12/app/innova2_flex_app -v
===============================================
 Verbosity:        1
 BOPE device:      None
 ConnectX device:  None
Cannot find appropriate ConnectX device
```
```
sudo ./mlxup 
Querying Mellanox devices firmware ...

Device #1:
----------

  Device Type:      ConnectX5
  Part Number:      SN37A48123_Ax
  Description:      ThinkSystem Mellanox Innova-2 ConnectX-5 FPGA 25GbE 2-port PCIe Adapter
  PSID:             LNV0000000025
  PCI Device Name:  /dev/mst/mt4119_pciconf1
  Base GUID:        aaaaaa0300aaaaa0
  Base MAC:         aaaaaaaaaaa0
  Versions:         Current        Available     
     FW             16.24.4020     N/A           
     PXE            3.5.0603       N/A           
     UEFI           14.17.0013     N/A           

  Status:           No matching image found
.... ( the same for device 2)
```

AGenchev, 2023-12-14:
Thank you so much for the invaluable information ! You really inspire me to go further. Finally I managed to obtain a Xilinx-DLC10 JTAG adapter, installed the recommended ubuntu, vitis+vivado & all the drivers. I'm not sure why I need all drivers if I'm not using the network card.
My card looks not fully functional right out of the box, probably the firmware is not OK.
[sudo lspci -vvvnnxxx output(https://gist.github.com/AGenchev/f466b047b2a9cd13bc1114208c3828d2#file-gistfile1-txt)]
I'm going to test courrupting firmware by shortening pins. Just wonder when and for how long the pins should be shorted. I'm going to solder wires to the flash pins.

AGenchev, 2023-12-15:
My next update: "A new hope"
Here is the prepared bed for the new W25Q128JVS IC
[image]
After my imperfect soldering job:
[image]

mwrnd, 2023-12-16:
> I'm not sure why I need all drivers if I'm not using the network card.

[`innova2_flex_app`](https://github.com/mwrnd/innova2_flex_xcku15p_notes#set-up-innova-2-flex-application) requires access to the first network interface to enable/disable JTAG and other features. The ConnectX-5 controls which [FPGA bitstream](https://github.com/mwrnd/innova2_flex_xcku15p_notes#initial-loading-of-the-flex-image) to run.

> how long the pins should be shorted

Until the driver for Recovery Mode is loaded. I fixed the notes [to make that more clear](https://github.com/mwrnd/innova2_flex_xcku15p_notes/commit/a51e6028b1aab01e52044ee572580dce51e5e5a7).

> Flint refuses to flash:
> -E- PSID mismatch. ... (LNV0000000025) differs ... (MT_0000000158)

Yes I recently tried the procedure on a board with a PSID different than `MT_0000000158` and had a similar experience. I was [eventually able to get it to work](https://github.com/mwrnd/innova2_flex_xcku15p_notes#try-overwriting-with-empty-image) using `mstflint`'s Write Block (`wb`) command.

> I decided to bypass the flint logic - ran the command
> and after it started reading shorted DO.
> ... Flash is write protected.

This is very difficult to get right as `flint` reads from the same addresses multiple times. My guess is that when it was unable to read back data it had written it assumed the IC is write-protected.

> After cold reboot, mlx drivers are still unhappy ...
> `Firmware over 120000 MS in pre-initializing state, aborting`

This one is new for me. It is my understanding that the W25Q128JVS gets loaded once at boot then resides in CX5 memory.

> Ran mlxup, after the following I gave it perm. to update to 16.28.2006

The ConnectX-5 works well enough to update its own firmware which is good news.

What is the [FBGA Code](https://github.com/mwrnd/innova2_flex_xcku15p_notes/tree/main/debug_notes#discerning-adlt-vs-adat-and-adit-variants) on your board's DDR4 ICs (D9TBK or D9WFR)?

[image]

What is your cooling solution? The [ICs get very hot very quickly](https://github.com/mwrnd/innova2_flex_xcku15p_notes#cooling-solution) which leads to all sorts of problems. Post a picture of your cooling setup.

> Could it be because the ports don't have SFP modules attached?

That would just make the board hotter. I run mine with no SFP modules or cables and it works fine.

> Should I put in your example guids:
> ... 0xc0dec0dec0dec0de ...

Did your board not include a label with the MAC IDs? use the values you saw when you first ran `flint query`. However, this should not matter.
```
Description:           UID                GuidsNumber
Base GUID:             aaaaaa0300aaaaa0        12
Base MAC:              aaaaaaaaaaa0            12
```

> Something prevents firmware from initializing?

Yes. I suspect it will take a while to figure out why.

What is the motherboard and CPU you are using? Does it support at least PCIe 3.0?

> The flex app also doesn't like what it sees:

Yes, without the ConnectX-5 Ethernet Controllers there is no way for the drivers needed by `innova2_flex_app` to load.
```
sudo mst start
cd ~/Innova_2_Flex_Open_18_12/driver/
sudo ./make_device
sudo insmod /usr/lib/modules/`uname -r`/updates/dkms/mlx5_fpga_tools.ko
lsmod | grep mlx
cd ~
sudo ~/Innova_2_Flex_Open_18_12/app/innova2_flex_app -v
```

What does `lspci` show?
```
sudo lspci -nn | grep "Mellanox\|Xilinx"
sudo lspci -tv | grep "0000\|Mellanox\|Xilinx"
```

[image]

> According your explanation, XDMA looks good for my case,
> but I'm still far from it (have to learn)

I have been working on a [Tutorial for XDMA](https://github.com/mwrnd/notes/tree/main/XDMA_Communication#creating-a-memory-mapped-xdma-block-diagram-design) and [XDMA Communication](https://github.com/mwrnd/notes/tree/main/XDMA_Communication#quickstart). [PCIe P2PDMA](https://www.kernel.org/doc/html/next/driver-api/pci/p2pdma.html) is a complex project I have not yet been able to get working.

AGenchev, 2023-12-16:
I wrote in chronological order; "..Flash is write protected." IMHO was caused by write protection indeed, because if you saw my first flint query, it said at the bottom: `Security Attributes:   secure-fw`. I thought they likely used the lock bits in the flash. I released the DO pin after flint agreed that there is nothing so it will flash it. It really failed to write there so I had to replace the flash with new.
> What is the FBGA Code on your board's DDR4 ICs (D9TBK or D9WFR)?

It seems you have nailed it:
[image]
It is for "ADIT" variant.
> What is your cooling solution?
   
I i would hardly call it solution, because I basically attached 2 m.board north-bridge radiators which have springs using elastic rubber band to hold them in place and to provide pressure on the springs. Against the smaller I put a blower fan, which provides enough airflow.
[image]
Yesterday I ran mlxup to update FW to 16.28.2006.
After this, the drivers are loading, firmware initialized O.K. Today we have a whole lot different picture:
> sudo lspci -tv | grep "0000\|Mellanox\|Xilinx"

```
-+-[0000:7f]-+-08.0  Intel Corporation Xeon E5/Core i7 QPI Link 0
 \-[0000:00]-+-00.0  Intel Corporation Xeon E5/Core i7 DMI2
             +-01.0-[03-06]----00.0-[04-06]--+-08.0-[05]----00.0  Mellanox Technologies Innova-2 Flex Burn image
             |                               \-10.0-[06]--+-00.0  Mellanox Technologies MT27800 Family [ConnectX-5]
             |                                            \-00.1  Mellanox Technologies MT27800 Family [ConnectX-5]
```

mwrnd, 2023-12-18:
> Flash is write protected ...
> `Security Attributes:   secure-fw`

That makes sense and thanks for pointing it out. I have not come across this.

> With FPGA power increased to 10 (what was that - overclock?)

Likely. My guess is that the FPGA runs some large DSP-heavy project.

> it went to 80 C

That still seems safe.

> All tests from innova2_flex_app fail (on my 4GB version).

Most fail for me on the 8GB ADLT variant as well. Luckily the tests have no relevance to the usefulness of the Innova-2.

> I think with 4GB it still has plenty of memory to experiment with.

Yes, 4GB is enough for everything you could try with these boards.

The 8GB ADLT has both SFP GbE interfaces connected to the ConnectX-5 with the intent of using PCIe P2PDMA for communication between the FPGA and Ethernet.

[image]

From the [4GB ADAT Product Brief](https://manualzz.com/download/33751280)(*manualzz.com*) it appears to have at least one direct FPGA SFP interface. This is good news but it means that the 8GB ADLT firmware is not designed for it. `innova2_flex_app` requires the 8GB ADLT firmware to be able to program the FPGA but it means the ConnectX-5 is not likely to be usable with the ADIT/ADAT variants. The 8GB ADLT firmware also chooses which FPGA Bitstream Image gets loaded; User, Flex, or Factory.

[image]

[Issue #3](https://github.com/mwrnd/innova2_flex_xcku15p_notes/issues/3) has more on the differences between the boards.

Does your board have any markings that indicate it is the AD**A**T or AD**I**T?

> I think with 4GB it still has plenty of memory to experiment with.
> Only lack of documentation can be a culprit.
> I will try to load your test for 8-bit DDR-4.

Yes, the [bitstream](https://raw.githubusercontent.com/mwrnd/innova2_ddr4_troubleshooting/main/test_adit_mt40a512m16/innova2_ddr4_8bit_byte-lane-0_ADIT_MT40A512_bitstream.zip) from [test_adit_mt40a512m16](https://github.com/mwrnd/innova2_ddr4_troubleshooting/tree/main/test_adit_mt40a512m16) will test 512MB of the 4096MB. I tried a straightforward port of [innova2_8gb_adlt_xdma_ddr4_demo](https://github.com/mwrnd/innova2_8gb_adlt_xdma_ddr4_demo) to the [MT40A512M16 (D9TBK)](https://www.micron.com/support/tools-and-utilities/fbga?fbga=D9TBK#pnlFBGA) but it fails. I need to run more tests to figure out why.

> Yesterday I ran mlxup to update FW to 16.28.2006.
> After this, the drivers are loading, firmware initialized O.K

Great news!

> For now, JTAG doesn't work for me ...
> so I'm using these steps:
> disable JTAG and schedule the Flex image,
> reboot,
> program with innova2_flex_app,
> reboot,
> and then test your loaded bitstream.

Yes, that is the correct sequence to program a Configuration Flash Memory ([MT25QU512 *RW193* ICs](https://www.micron.com/support/tools-and-utilities/fbga?fbga=RW193#pnlFBGA)) bitstream using `innova2_flex_app`. This is what gets loaded during system boot. It is best to disconnect or at least power off your JTAG adaptor when using `innova2_flex_app`.

Using [JTAG to load a bitstream into the FPGA's Temporary SRAM Configuration Memory](https://github.com/mwrnd/innova2_xdma_demo#loading-updated-designs-using-jtag) is useful for development but not required.

> 05:00.0 Class [2000]: Mellanox Technologies Innova-2 Flex Burn image [15b3:0264]

It looks like your FPGA Flash Configuration Memory already includes the latest Flex Image. What version does `innova2_flex_app` say it is at?

[image]

> should I replace the MOSFET

No, your board seems to work. I had clear damage to my board.

> For now, JTAG doesn't work for me ...

After `05:00.0 Memory controller: Xilinx Corporation Device 9038` shows up, run `innova2_flex_app` to Enable JTAG. Does that help? Are you able to [use XSDB to communicate with the FPGA](https://github.com/mwrnd/innova2_xdma_demo#program-the-fpga-using-jtag)?

[... 20 lines cut ...]

```
cd ~/dma_ip_drivers/XDMA/linux-kernel/xdma/
sudo make uninstall
```
```
cd ~
wget https://codeload.github.com/Xilinx/dma_ip_drivers/zip/a93d4a4870e41d152b33aebb3f869eefb11aa691 -O dma_ip_drivers-a93d4a4.zip
unzip dma_ip_drivers-a93d4a4.zip
mv dma_ip_drivers-a93d4a4870e41d152b33aebb3f869eefb11aa691 dma_ip_drivers

cd ~/dma_ip_drivers/XDMA/linux-kernel/xdma/
make DEBUG=1
sudo make install

sudo depmod -a
sudo ldconfig

cd ~/dma_ip_drivers/XDMA/linux-kernel/tools
make

sudo reboot
```

Do you have access to a [large ultrasonic cleaner](https://www.google.com/search?q=ultrasonic+electronics+cleaner), a well-ventilated space such as a balcony or garage, and [99% Isopropyl Alcohol](https://mgchemicals.com/category/electronics-maintenance/isopropyl-alcohol/)? It has been my experience that old server equipment that is flaky becomes usable after about a half hour of ultrasonic Isopropyl cleaning.

AGenchev, 2023-12-19:
> What version does innova2_flex_app say it is at?

`*** FPGA image version:   0xc1 `
>  ...but it means the ConnectX-5 is not likely to be usable with the ADIT/ADAT variants. 

my goal is to use  ConnectX-5 as .. just a PCI-E bridge to allow the FPGA on the PCI-E bus.
I succeeded in running your test at the point "I ran a sanity check and tried sending 1GB of data. Only 512MB transfers as expected."
```
sudo ./dma_from_device --verbose --device /dev/xdma0_c2h_0 --address 0x0 --size 1073741824 -f recv.raw  
dev /dev/xdma0_c2h_0, addr 0x0, aperture 0x0, size 0x40000000, offset 0x0, count 1
host buffer 0x40001000, 0x7fe0b4b6d000.
/dev/xdma0_c2h_0, read underflow 0x20000000/0x40000000 @ 0x0.
#0: underflow 536870912/1073741824.
#0: CLOCK_MONOTONIC 10.724906794 sec. read 536870912/1073741824 bytes
```
The results are repeatable, xdma works. The speed is ~ 50 MB/s when 512 MB of 1 GB are transferred.
Now I guess the unknown is how the memory is connected to the FPGA (which pin goes where). 
The JTAG doesn't run as expected. The last test I did was to enable JTAG on the APP and then to run xsdb% targets 
but nothing shows:
```
tcfchan#0
xsdb% after 7000                                                                                                                        
xsdb% targets                                                                                                                           
  1  whole scan chain (board power off)
xsdb% target 1                                                                                                                          
xsdb% fpga -state                                                                                                                       
No supported FPGA device found
```

mwrnd, 2023-12-20:
>> the bitstream from test_adit_mt40a512m16
> I flashed it and succeeded in running your test ...

So the checksums match for 512MB transfers?

More and more of the board is proving to work. Let's hope the success continues.

> speed is ~ 50 MB/s when 512 MB

Please post the output of `sudo lspci -nnvv` for the Mellanox and Xilinx devices to confirm the links are all at the correct Width and Speed and whether there have been any [RX and/or TX errors](https://superuser.com/questions/915430/the-lspci-vvv-output-correrr-and-uncorrerr-changes-over-time).

> 72-bit bus E.g. pin "constraints" are not known - am I right ?

[Constraints for the 8GB ADLT are known](https://github.com/mwrnd/innova2_8gb_adlt_xdma_ddr4_demo/blob/eb20ab06cd3bfbe08d2601a7ba6bb28e7f18ce93/innova2_constraints.xdc#L122). When I use the same constraints in [innova2_8gb_adlt_xdma_ddr4_demo](https://github.com/mwrnd/innova2_8gb_adlt_xdma_ddr4_demo) but change the *Memory Part* of the DDR4 IP Block to `MT40A512M16LY-075` and load it into the 4GB ADIT I get incorrect data reads. When I then connect to the 4GB ADIT board via [JTAG and access *MIG_1*](https://github.com/mwrnd/innova2_ddr4_troubleshooting) I get `MIG Status: PASS`.

If you have the time, try [recreating `innova2_8gb_adlt_xdma_ddr4_demo`](https://github.com/mwrnd/innova2_8gb_adlt_xdma_ddr4_demo#recreating-the-design-in-vivado) but change the DDR4 Block to [use `MT40A512M16LY-075` and the memory range to 4GB](https://github.com/mwrnd/innova2_ddr4_troubleshooting/blob/main/test_adit_mt40a512m16/README.md). It may be that my board has faulty memory ICs.

> The JTAG doesn't run as expected.

Are you able to communicate with other Xilinx FPGAs using your debugger?

I am on my second [14-Pin JTAG cable](https://www.trustedparts.com/en/search/6003-310-001)([1](https://digilent.com/shop/jtag-2x7-ribbon-cable/)). With everything powered off, check with a multimeter in continuity mode that all the JTAG signals connect between the Innova2 and your JTAG adapter and are not shorted to GND.

AGenchev, 2023-12-21:
Thank you for the guidance! I will be able to do all this after 7th JAN 2024, because I am on vacation far from home till then w/o access to IT equipment.

mwrnd, 2024-01-08:
I got DDR4 working on my Innova2 4GB ADIT board. The [`innova2_4gb_adit_xdma_ddr4_demo`](https://github.com/mwrnd/innova2_4gb_adit_xdma_ddr4_demo) project includes bitstreams and instructions for testing.

My problem turned out to be a reset issue. When I tried porting the [`innova2_8gb_adlt_xdma_ddr4_demo`](https://github.com/mwrnd/innova2_8gb_adlt_xdma_ddr4_demo/tree/eb20ab06cd3bfbe08d2601a7ba6bb28e7f18ce93) project to the 4GB ADIT I also moved from Vivado 2023.1 to 2023.2 which changed how resets worked. I carefully updated the reset network and it now works and should be more robust.

AGenchev, 2024-01-08:
>I got DDR4 working on my Innova2 4GB ADIT board. The [innova2_4gb_adit_xdma_ddr4_demo](https://github.com/mwrnd/innova2_4gb_adit_xdma_ddr4_demo) project includes bitstreams and instructions for testing.

This is really good news ! I'll try it. I only found time to test the JTAG adapter vs another board (an old NETFPGA-1G-CML) and it worked. This means I have also kind of problematic JTAG port on the innova2.

AGenchev, 2024-01-10:
Now I'm trying to understand some basics: the block scheme of 4gb_adit_xdma_ddr4_demo looks rather complex, so it seems it does more than just being a XDMA to int.DDR4 memory controller. I see for the memory you set a different arbitration scheme (RD PRI REG) on the ADIT board while on ADLT. I have to read much more to get what is going on. I think to create a very basic block design using the DDR-4 guidelines from your demo, then to add to it more blocks trying to generate something into DDR4 and read it. Not sure if 2 devices try to access the DDR4 slave AXI at the same time what will happen.

mwrnd, 2024-01-11:
> 4gb_adit_xdma_ddr4_demo looks rather complex

Yes, I added various blocks to [test other parts of the board](https://github.com/mwrnd/innova2_4gb_adit_xdma_ddr4_demo/tree/main#custom-software-for-accessing-axi-blocks).

> I have installed [Vivado] 2021 ...
> basic block design using the DDR-4 guidelines from your demo

Yes, a much simpler design should work in Vivado 2021.2 to test DDR4:

[image]

I plan to eventually update my `innova2_flex_xcku15p_notes` notes for Vivado 2023.2 and Ubuntu 22.04.

> you set a different arbitration scheme (RD PRI REG)

The AXI Arbitration Scheme should not matter for such a simple design. For designs with multiple AXI blocks that use DDR4 Round-Robin guarantees low-latency access at the cost of throughput. To be honest I forgot to change it from the default [Read Priority](https://support.xilinx.com/s/question/0D52E00006hpg0lSAA/mig-22-question?language=en_US). Refer to the [Memory IP Guide (pg150) Pg#143](https://docs.xilinx.com/v/u/en-US/pg150-ultrascale-memory-ip).

> xcku15p doesn't come with free license

Yes, unfortunately not. You can use an evaluation license or the [AWS Vivado 2023.2 AMI](https://aws.amazon.com/marketplace/pp/prodview-xwxhdnynrshyc) by the hour.

> JTAG adapter vs another board (an old NETFPGA-1G-CML) and it worked.
> This means I have also kind of problematic JTAG port on the innova2.

The Innova-2 uses 1.8V for JTAG. Does your adaptor support that? I recommend one of the [Officially-Supported JTAG Adaptors](https://docs.xilinx.com/r/en-US/ug908-vivado-programming-debugging/JTAG-Cables-and-Devices-Supported-by-hw_server) or the [Waveshare Platform Cable clone](https://www.waveshare.com/platform-cable-usb.htm). I am aware of inexpensive FT2232H-based JTAG SMT boards [causing problems](https://forums.servethehome.com/index.php?threads/identifying-some-curious-xilinx-solarflare-cards.35111/page-3#post-377463).

> Why I'm trying to get JTAG working - to skip reboot
> when I want to set a new image to the FPGA.

Working JTAG will also allow you to use [XSCT/XSDB](https://docs.xilinx.com/r/en-US/ug1400-vitis-embedded/XSCT-Commands) to [debug soft-core processors like the RISC-V](https://github.com/mwrnd/innova2_experiments/tree/main/riscv_rocket64b4l2w_xdma) or MicroBlaze.

AGenchev, 2024-01-12:
I report JTAG success ! The JTAG adapters I tried are on the picture.
[image]
First one I bought for US$17 on Aliexpress - it is a Chinese clone. The second one costs here $300 and I don't own it. Counterintuitive, the cloning performs slightly faster. You were right - my JTAG was *not* broken. I don't know now the reason it didn't work - very likely my misunderstanding the usage of **innova2_flex_app**:
When you set the FPGA image to boot, you need to reboot the computer. So probably I inferred that I need to enable the JTAG, then reboot the computer so it boots with JTAG enabled. When the cheap DLC9 arrived (yesterday) I observed that it lacks suitable connector, so I used the ribbon cable from DLC10 with it just to test whether it works, then decided to experiment without rebooting the computer on the innova2. And boom - it worked. Then tested DLC10, it worked as well. 
Now I'm implementing the design you gave me above. After Vivado2023.2 (a new web install) killed the OS (OOM) with just 2 processes on 32GB RAM, i installed and configured zram and I it helped a lot.

mwrnd, 2024-01-14:
> I report JTAG success !

Excellent news!

> misunderstanding the usage of innova2_flex_app

Thanks for pointing that out. I have [tried to improve my notes].(https://github.com/mwrnd/innova2_flex_xcku15p_notes/commit/c8ff37c2ea6f5e4ffe4256aa38bb74c110bf4ae0).

> After Vivado2023.2 (a new web install) killed
> the OS (OOM) with just 2 processes on 32GB RAM

When you run `top` do you have a [swap partition](https://help.ubuntu.com/community/SwapFaq)? Vivado somehow manages to always get into swap even if a system has plenty of memory. Try the following which makes swap (Virtual Memory) [less likely to be used](https://communities.vmware.com/t5/VMware-Workstation-Pro/VMWare-workstation-in-a-fistfight-with-Linux-Memory-Compactor/td-p/2876992). It is issues like these that make me wary of updating my notes for the latest Vivado too soon.

```
sudo su
echo 0 > /proc/sys/vm/compaction_proactiveness
echo never > /sys/kernel/mm/transparent_hugepage/enabled
echo never > /sys/kernel/mm/transparent_hugepage/defrag
echo vm.compaction_proactiveness=0 >>/etc/sysctl.conf
sysctl -w vm.compaction_proactiveness=0
sysctl -w vm.extfrag_threshold=1000
exit
```

> succeeded and of curiosity overclocked the DDR4 to 2666. retention still OK

Good to know. I have [added a comment](https://github.com/mwrnd/innova2_4gb_adit_xdma_ddr4_demo/commit/0cb05f2f8573d78a505aee922d37710752651077) to the [innova2_4gb_adit_xdma_ddr4_demo](https://github.com/mwrnd/innova2_4gb_adit_xdma_ddr4_demo) project.

> Shall i close this issue, because we solved everything and I went off-topic

Yes, please do. Create new issues if you encounter any problems.

closed 2024-01-14 by AGenchev

AGenchev, 2024-01-14:
> When you run top do you have a [swap partition](https://help.ubuntu.com/community/SwapFaq)?

Of course, but I had only 1 GB, because even though I use a PMR server HDD w 256MB cache, 32G swap would be awfully slow.
Thanks for kernel config ideas, before try them, I ran this script (for 32GB RAM):
```
#!/bin/sh
swapoff -a
modprobe zram
zramctl /dev/zram0 --algorithm zstd --size 32G
mkswap  /dev/zram0
swapon --priority 100 /dev/zram0
swapon -a
swapon -s
```
And top now shows:
```
MiB Mem :  32039,7 total,  16255,2 free,  11414,2 used,   4370,2 buff/cache
MiB Swap:  33744,0 total,  33744,0 free,      0,0 used.  18705,4 avail Mem
```
(Vivado is running ob background)