mwrnd/innova2_flex_xcku15p_notes #9: How to check the board HW if the jtag is not working

hzy199411, 2024-07-20:
I have one innova2 flex board, when I use the innova2_flex_app enable the jtag access, but actually the jtag scan is failed.
```
open_hw_target
INFO: [Labtoolstcl 44-466] Opening hw_target localhost:3121/xilinx_tcf/Digilent/C30593ECABCD
ERROR: [Labtools 27-2269] No devices detected on target localhost:3121/xilinx_tcf/Digilent/C30593ECABCD.
Check cable connectivity and that the target board is powered up then
use the disconnect_hw_server and connect_hw_server to re-register this hardware target.
ERROR: [Common 17-39] 'open_hw_target' failed due to earlier errors.
```

so I suspect there maybe some HW broken. 

I found here is a case that U41 and U49 component may cause the jtag not works in https://github.com/mwrnd/innova2_flex_xcku15p_notes/tree/main/debug_notes#debug-jtag-components

can you please help share more detail about the U41 and U49?
I measured the voltage information are:
| U41| Voltage | Diode Mode Value |
| - | - | - |
|PIN D| 1.18V | 443 |
|PIN G| 0V | 466 |
|PIN S| 0V | 406|

| U49| Voltage | Diode Mode Value |
|PIN D| 3.3V | 553 |
|PIN G| 0V | 462 |
|PIN S| 0V | 0 |

is the voltage and Diode mode value correct?

and can you please share a clear bottom picture for reference, I want to check is there some component missed.

mwrnd, 2024-07-27:
I am reading the same voltages on a powered up (in-system) board with no JTAG connected.

Please confirm your JTAG runs at 1.8V and [works with Xilinx parts](https://docs.xilinx.com/r/en-US/ug908-vivado-programming-debugging/JTAG-Cables-and-Devices-Supported-by-hw_server). If possible, please test your JTAG on a known-working Xilinx FPGA board.

[image]

> so I suspect there maybe some HW broken.

When you power up your system, which LEDs on the Innova-2 are on?

> I use the innova2_flex_app enable the jtag access

Is your [CX-5 firmware the newest](https://raw.githubusercontent.com/mwrnd/innova2_flex_xcku15p_notes/main/img/Flex_Image_Version.jpg)? Please post the output of `innova2_flex_app`.

[image]

On a **not powered**, (unplugged, out-of-system) board with a multimeter in Diode Mode:

[image]

I measured the transistors:

[image]

The following are my measurements:

[image]

> please share a clear bottom picture for reference

https://github.com/mwrnd/innova2_flex_xcku15p_notes/blob/main/img/Innova-2_Flex_8GB_XCKU15P_MT27808_MNV303212A-ADLT_Back.jpg
https://github.com/mwrnd/innova2_flex_xcku15p_notes/blob/main/img/Innova-2_Flex_4GB_XCKU15P_MT27808_MNV303212A-ADIT_Back.jpg
https://github.com/mwrnd/innova2_flex_xcku15p_notes/blob/main/img/ADLT_vs_ADIT_Back-of-PCB_Comparison.jpg

hzy199411, 2024-07-28:
@mwrnd Thanks for your information, 
today I found the reason is that my board can only support to using the jtag at 12MHz. if I reduce the clock setting to 10M using the JtagSmt2 or use the 12M if using the DLC10 cable.

update some information about why I try to reduce the clock.
first I use the "djtagcfg init -d JtagSmt2" to detect the jtag and it can found the jtag ID but it report unknown device:
```
$ djtgcfg init -d JtagSmt2
Initializing scan chain...
Found Device ID: 14a56093

Found 1 device(s):
    Device 0: UNKNOWN
```

then I measured the jtag waveform when run the command and confirmed the waveform result is good.
(the UNKNOWN device maybe that the digilent adept utilities don't support the KU devices).

but when I measured the jtag waveform when open Vivado HW manager and the waveform report many 0xFF 0xFE error data.
I compared the clock setting and found the clock frequency is higher than the case using "djtagcfg init -d JtagSmt2" command.
so I suspect the clock maybe too fast, then I decrease the clock to 10M and found that the jtag is detect.

can you help check is your board support to use higher jtag clk frequency such as 15M?
I want to is the 15M jtag clock not working issue only happens on my board or other boards.

closed 2024-08-22 by hzy199411